NemoAlex/openwrt-wrt1900ac-docs #7: any chance for an english version?

glyph, 2015-12-25:
After reading https://translate.googleusercontent.com/translate_c?depth=1&hl=en&ie=UTF8&prev=_t&rurl=translate.google.com&sl=auto&tl=en&u=https://github.com/NemoAlex/openwrt-wrt1900ac-docs&usg=ALkJrhj5HJQupWXK0UE6_mh4LSzhHi7vzA it and seeing some of your comments on openwrt issues it became clear to me that this is probably the easiest place to get updated wifi drivers for the wrt1900ac.  But I don't speak chinese :).  Google seems to have done a reasonable job with the english translation, but it would still be nice to have a few `.en.md` files for those of us who have these routers but not the language skills.

If not, I understand!  Thank you for doing this work.

NemoAlex, 2015-12-25:
I was a little surprised to got some attention.
The main purpose of this repository is to provide some tools such as ShadowSocks and ShadowVPN, which are useful for Chinese users. ([Because of the GFW](https://en.wikipedia.org/wiki/Great_Firewall))
The mwlwifi driver is only a part of that.
So I'm considering to make a new repository to follow the mwlwifi project. I'll write a English readme for that.
Thanks for your proposal!

NemoAlex, 2015-12-25:
Need tag and reorganize many versions.. require some time.
I'll finish it tomorrow.

glyph, 2015-12-25:
I ended up here after looking at your comments on https://github.com/kaloz/mwlwifi/issues/21.  Those of us with random crashes on our wrt1900ac's are looking for something to do, and setting up an openwrt build environment is a lot trickier and more difficult than just downloading a file.  I have no idea if it will fix things, but I will let you know if I still experience crashes :).

NemoAlex, 2015-12-31:
Hi @glyph
Here is the project: https://github.com/NemoAlex/mwlwifi-bin
Sorry for waiting so long. Have been really busy these days.

closed 2015-12-31 by NemoAlex

glyph, 2016-01-03:
> Sorry for waiting so long. Have been really busy these days.

No need to apologize - I really appreciate you doing this!