NemoAlex/openwrt-wrt1900ac-docs #8: crash after installing 20151216 wifi driver

glyph, 2015-12-25:
I downloaded `kmod-mwlwifi_3.18.20+10.3.0.15-20151216-1_mvebu.ipk` and did `ipkg install`, then `rmmod mwlwifi ; modprobe mwlwifi` and my router crashed. When it came back up, it did not load wifi drivers.

NemoAlex, 2015-12-25:
The right way to upgrade mwlwifi is do  `opkg install kmod-mwlwifi_3.18.20+10.3.0.15-20151216-1_mvebu.ipk`, then reboot your router, done.

I'm not sure what will happen if you do these steps above.
Probably you need reinstall the package and reboot it.

Also please note the release version of OpenWrt. Softwares in this respository only works in OpenWrt 15.05 final.

glyph, 2015-12-25:
Yes, I had the same problem with 20151208, and then I tried rebooting, and everything worked fine. The other problems I was having were filesystem inconsistencies resulting from installing the package and not doing a `sync` before it crashed.

Thanks for your response.

glyph, 2015-12-25:
By the way: I am indeed using the release version of 15.05.

closed 2015-12-25 by glyph