NemoAlex/openwrt-wrt1900ac-docs #9: 5g wifi slow issue

billzhuang, 2016-01-08:
谢谢你做了这些工作，我就偷懒直接用在我1200ac上。

我发现wifi会越来越慢，reboot后就恢复。

怎么排查这个问题？

NemoAlex, 2016-01-08:
如果你用的是 release 里的集成固件的话：
查看系统日志，是否有什么报错
检查内存占用

有些问题是由于无线驱动的不完善导致的，社区对此也无能为力，只能等作者更新，改进驱动。
这两天我会做一个新的版本，集成新驱动，Samba 的稳定性会有所提升。

billzhuang, 2016-01-08:
那请问你遇到了ssh很卡的问题么？
这个issue，https://dev.openwrt.org/ticket/21446#comment:2

cpu, io, memory都没问题，系统日志没仔细看，扫了下没啥问题。

NemoAlex, 2016-01-08:
没有遇到过这种速度这么低的情况。
可能是个别客户端和它的兼容性问题？
ssh 卡，在过去的版本里比较常见，最近版本的驱动（例如 Nemo 1.1 集成的）已经没有再发现了。

billzhuang, 2016-01-08:
好的，我决定重刷一遍，然后更新下你最近push的驱动看看~

谢谢你啦啦啦

NemoAlex, 2016-01-08:
注意：集成固件的 1.1 版和官方的 15.05 不兼容，所以不能安装那些包来更新驱动。0.1 版可以。
接下来计划的 1.2 版也会兼容。

billzhuang, 2016-01-08:
好凶险，所以我的姿势是：
1. 安装0.1版
2. 更新这里的ipk https://github.com/NemoAlex/openwrt-wrt1900ac-docs/tree/master/ipks/chaos_calmer-15.05-mvebu
3. 更新这里的驱动https://github.com/NemoAlex/mwlwifi-bin

对不？

NemoAlex, 2016-01-08:
可以的。或者等两天，1.2 版也会出炉。

billzhuang, 2016-01-08:
那我等你吧，辛苦啦。

closed 2016-01-08 by billzhuang

billzhuang, 2016-01-12:
请教个问题，为何1.1版只是集成了些apk，为啥会和官方以及对应的wifi驱动的ipk包不兼容，这个不兼容是怎么引入的？好奇

NemoAlex, 2016-01-12:
0.1 版使用了官方编译的内核和主要模块，而 1.1 版是完全重新编译的。
确实，1.1 版对源码有过改动。加入了 WRT1900ACS 的兼容，以及默认开启 Wifi 之类的东西。所以会导致带有 kmod- 开头的包不兼容。
为了实现深入订制，第三方订制的固件普遍采用了重新编译的这种方式，甚至内核版本都和官方的稳定版不同。从而导致和官方软件源完全不兼容。在这一点上，1.1 上已经尽力去兼容了，但是毕竟有限。
以后的版本还会延续 0.1 版的方式，只集成稳定和质量可靠的开源软件。

billzhuang, 2016-01-12:
了解，谢谢~

NemoAlex, 2016-01-12:
新固件发布了，可以试一下
http://koolshare.cn/thread-24139-1-1.html

billzhuang, 2016-01-12:
刷了新版，问题消失了，哈哈哈哈哈哈哈啊哈

在观察两天，之前一般是5分钟之后就挂，哈哈哈哈哈哈哈

billzhuang, 2016-01-13:
一夜过来，5g速度下降10%，继续观察。
相较上个版本，10分钟就下降一半好太多。
晚上回去再看下

billzhuang, 2016-01-13:
报告下，今晚回家5g 网速已经drop到三分之一左右，看了kernal log，显示
```
[43212.787947] ieee80211 phy0: check ba result error 1
[43212.792846] ieee80211 phy0: ampdu start error code: -22
[43212.827947] ieee80211 phy0: check ba result error 1
[43212.832847] ieee80211 phy0: ampdu start error code: -22
[43212.867957] ieee80211 phy0: check ba result error 1
[43212.872857] ieee80211 phy0: ampdu start error code: -22
[43212.907944] ieee80211 phy0: check ba result error 1
[43212.912844] ieee80211 phy0: ampdu start error code: -22
```

和这个bug一致https://github.com/kaloz/mwlwifi/issues/41

dawncold, 2017-01-17:
@billzhuang 我曾经遇到ssh很卡并且在5G wifi中vim openwrt上的文件时断线情况，但用2.4G没有问题。

billzhuang, 2017-01-20:
@dawncold 然后你咋解决的？换LEDE?

dawncold, 2017-01-20:
@billzhuang 我的wrt1900ac是美版的，当时5G用auto channel，怀疑是不是和家中其他设备不太兼容，现在固定149或者157，刚刚试了下没重现，你可以试试看。

shao222, 2017-01-20:
请问你们怎么测试5G信号的？

发自我的手机

dawncold, 2017-01-20:
没有很严格测试，只是当时在用5G网络时发现很慢或者容易断，ping网关也不太稳定，有个工具叫wavemon可以测测