NemoAlex/openwrt-wrt1900ac-docs #11: 大大您好，能否为这个固件添加单线并发多拨的选项？

jun9100, 2016-01-25:
大大您好，能否为这个固件添加单线并发多拨的选项？
之前是用这位大大的固件，自己的机器是wndr4300，现在换了新设备linksys 1900能否百忙之中帮忙解决下。十分感谢。
http://www.right.com.cn/forum/thread-146171-1-1.html

NemoAlex, 2016-01-25:
目前没有什么兴趣研究多播。你可以自己弄，这个固件支持官方源，装东西还是很方便的。针对其它机器的教程也可以参考。
On 周一, 1月 25, 2016 at 11:22 下午, NemoAlex/openwrt-wrt1900ac-docs
<reply@reply.github.com>
wrote:
大大您好，能否为这个固件添加单线并发多拨的选项？
之前是用这位大大的固件，自己的机器是wndr4300，现在换了新设备linksys 1900能否百忙之中帮忙解决下。十分感谢。
http://wwwrightcomcn/forum/thread-146171-1-1html
[http://wwwrightcomcn/forum/thread-146171-1-1html]

—
[https://github.com/NemoAlex/openwrt-wrt1900ac-docs/issues/11] .[https://github.com/notifications/beacon/ABWJhOQPp4CWBbeJRijqANJsCAaYS1V8ks5pdjVBgaJpZM4HLsS9.gif]

jun9100, 2016-01-25:
感谢您的回复，谢谢。

jun9100, 2016-01-29:
在网上搜索了相关的教程，感觉都太老了，大大能否提供一个可参考的？

nadoo, 2016-01-31:
创建macvlan类型的虚拟接口成功，虚拟接口上pppoe拨号始终不成功，所以无法多播，不知其他人有这个情况吗？

NemoAlex, 2016-02-01:
@nadoo 
https://github.com/NemoAlex/openwrt-wrt1900ac-docs/issues/2

jun9100, 2016-02-02:
@nadoo 这位兄弟你搞定多拨了吗？

NemoAlex, 2016-02-02:
http://koolshare.cn/thread-31083-1-1.html
看来要有教程了

jun9100, 2016-04-19:
感谢分享

griffinqiu, 2016-05-03:
多拨+1  [koolshare的教程](https://koolshare.cn/forum.php?mod=viewthread&tid=31138&page=1&extra=#pid229567) 有点晦涩。 仅仅不同op版本的pppd 的编译就将无数新手档在门外。

griffinqiu, 2016-05-06:
@NemoAlex 他[这个教程](https://koolshare.cn/forum.php?mod=viewthread&tid=31138&page=1&extra=#pid229567)说的pppd在15.05.1上应该怎么编译？这个编译出来似乎不是ipk

Rabbit52, 2016-06-02:
我参考的这个教程，多播成功了但是我这不能叠加下行，就没玩了
http://ju.outofmemory.cn/entry/214459

jun9100, 2016-06-02:
电信目前都只能叠加上行了

发自我的 iPhone

griffinqiu, 2016-06-03:
@Rabbit52 不行，我按照这个弄完了，但是新加的不能拨号，之前的网件都可以多拨。 
@jun9100 就是要上传；）  你的光猫的wan口是不是不是千兆的？

jun9100, 2016-06-03:
我现在已经没用1900ac了，我搞了个软路由。之前wndr3800和4300都可以多播叠加上传的。

发自我的 iPhone

opchen, 2017-11-25:
最近买了一个Linksys WRT1200AC，翻阅了很多文章，最终解决多拨问题。
其实只是需要将macvlan虚拟出来的网卡设置为混杂模式即可，其余按照标准操作即可，参考以下命令：
`ifconfig $ifwan promisc`

希望对你们有帮助