NemoAlex/openwrt-wrt1900ac-docs #16: readme有点点没说清楚

griffinqiu, 2016-04-13:
WRT1900AC有两个Flash。Flash-A，Flash-B。

在Flash-A下升级或者安装固件会刷新Flash-B的固件。反之亦然。

所以升级OpenWRT最安全的做法是切换到另外一个Flash进行安装。 这样可以降低刷成砖的风险。 如果一直这样做可以保证一个原厂的Flash固件和一个OpenWRT固件共存。

切换方法： 开启路由器后等3秒关闭。反复四次。

NemoAlex, 2016-04-13:
其实我觉得不切也问题不大，因为既然能刷过去，说明之前的系统是完整可运行的，即便另一个分区挂了，3次启动不成功也会自动切回原来的分区。

griffinqiu, 2016-04-13:
嗯你说的对。
请问这个教程是最新稳定版本吗？
https://github.com/NemoAlex/openwrt-wrt1900ac-docs/wiki/%E7%AE%80%E6%98%93%E6%8C%87%E5%8D%97%EF%BC%9A%E4%BD%BF%E7%94%A8-OpenWrt-%E7%9A%84%E4%BA%A4%E5%8F%89%E7%BC%96%E8%AF%91-SDK-%E6%9D%A5%E7%BC%96%E8%AF%91-ipk-%E8%BD%AF%E4%BB%B6%E5%8C%85

closed 2016-04-13 by griffinqiu

NemoAlex, 2016-04-13:
是的，只不过现在是 15.05.1 了，把 SDK 的地址换一下就可以了。

griffinqiu, 2016-04-13:
嗯， 之前都是用别人编译好的。
一般默认要编译上那些软件比较够用？有没有你自用的软件列表作为参考？

NemoAlex, 2016-04-13:
这里有一个简单的列表
http://koolshare.cn/thread-39880-1-1.html
其实你用这个集成版就行，不需要的软件可以再删。

griffinqiu, 2016-04-13:
wow， 你发的这个链接太有用了。 多谢！

griffinqiu, 2016-04-13:
给人以鱼不如授之以渔。赞一个！！！

griffinqiu, 2016-04-13:
我现在用的SS + UDP Forward 的黑名单模式。 之前试过还ChinaDNS。有时候会出现解析不到国外域名的情况。
请问现在最合适的分流方式是那种？

NemoAlex, 2016-04-13:
ChinaDNS 并不好用。稳定的方式还是域名黑名单 + IP段黑名单/白名单。

shao222, 2016-04-13:
@nemo 你所说的有现成的方案吗？

griffinqiu, 2016-04-13:
@shao222 有，这里。
https://cokebar.info/archives/962

shao222, 2016-04-13:
@GriffinQiu 感谢 我去研究一下。

griffinqiu, 2016-04-13:
@NemoAlex 再问一个问题嘛
升级OpenWRT你一般备份&还原那些文件？我之前每次升级都是重做一般。备份/etc下面相关的文件似乎是一个方法。 可不可以看一下你的Backup file list。
另外跨OpenWRT版本的配置备份可以直接用吗？

3Q

NemoAlex, 2016-04-13:
重新配置一遍比较好，因为保留的配置有可能不适用于新的系统或者软件，容易造成问题。

griffinqiu, 2016-04-13:
哦， 小版本的sysupgrade会不会覆盖配置呢？