NemoAlex/openwrt-wrt1900ac-docs #17: 请问1.3能默认不使用SS代理吗？

griffinqiu, 2016-04-15:
我想自己用ipset来指定黑名单域名的IP走SS。默认不走SS。

griffinqiu, 2016-04-15:
我通过下面下面几步解决了这个问题
1. 编译了shadowsocks-libev_2.4.5-1_mvebu.ipk。
2. 删除shadowsocks-libev-spec_2.4.5-1_mvebu.ipk。
3. 按照[这篇文章](https://cokebar.info/archives/962) 的方案二配置

现在可以实现我指定的域名走SS，默认走电信。 同时我产生了几个问题：
1. shadowsocks-libev和shadowsocks-libev-spec可以共存吗？ 安装shadowsocks-libev的时候报错ss-redir等文件文件shadowsocks-libev-spec已经安装了。
2. shadowsocks-libev-spec模式下能不能实现默认走电信，按照[gfwlist2dnsmasq](https://github.com/cokebar/gfwlist2dnsmasq)匹配后走SS。 有这种需求的原因是我有一台NAS，平时都在下东西，P2P的下载世界各地的IP都有可能，同时用的SS服务器是按流量收费的，害怕流量费过高
3. shadowsocks-libev有luci吗 ；）

NemoAlex, 2016-04-15:
1. 应该是不可用共存
2. spec 只有白名单的模式，也就是 chinadns_chnroute.txt
3. 官方没有，但是我看有人做这种的，不知道源码在哪

griffinqiu, 2016-04-15:
“有这种需求的原因是我有一台NAS，平时都在下东西，P2P的下载世界各地的IP都有可能，同时用的SS服务器是按流量收费的，害怕流量费过高”
你有没有碰到过我这种需求， 用shadowsocks-libev-spec有没有可能解决呀？

griffinqiu, 2016-04-15:
我考虑过这种方案。但是还是避免不了其它设备不经意的情况下从国外下载大量的数据。阿里云的流量特贵了。5毛钱一个G；（

closed 2016-04-15 by griffinqiu

griffinqiu, 2016-04-16:
@NemoAlex 请问你说的这个方案应该用哪种方法来防止DNS污染呢

NemoAlex, 2016-04-16:
如果不用 ChinaDNS 的话，可以 dnsmasq 设置某些域名的上游使用国外 DNS，前提是你的 SS 服务器支持 UDP。