NemoAlex/openwrt-wrt1900ac-docs #19: 你们无线传送速度都是多少呀

griffinqiu, 2016-04-18:
我用RMBP测试传NAS里的东西。NAS是有线连接路由器。
5GHz下，路由器重启后最高可以到达40m/s， 但是持续得到2分钟，后面就一直是7m/s-10m/s。
2.4GHz下，最高就7m/s。
反正每次路由器重启后速度就很快，一会儿就不行了

griffinqiu, 2016-04-19:
降级到10.3.0.15，5GHz稳定在20-30mb/s

griffinqiu, 2016-04-19:
10.3.0.17很稳定啊，不掉速。 能隔一堵墙稳定在30左右

NemoAlex, 2016-04-19:
@griffinqiu 我试验的过程中遇到了很多次掉线的情况，不知道是不是个例

griffinqiu, 2016-04-20:
@NemoAlex 有时候刚刚启动的时候经常掉线。 每个版本都有。 
2.4G每个版本都不是很稳定呀

NemoAlex, 2016-05-03:
@griffinqiu 一直在用.17吗？现在怎么样？

griffinqiu, 2016-05-04:
@NemoAlex .17很稳定, 兄弟!

griffinqiu, 2016-05-04:
比 .15的速度快, 基本上不怎么掉速度

NemoAlex, 2016-05-04:
@griffinqiu 想问一下是 20MHZ？有没有试过 80MHZ？
我测试 .17 在 80MHZ 的时候非常慢。而其它版本的驱动都不会。

shao222, 2016-05-04:
我好像也更新到 .17 了，而且也是80Mhz 但是速度一直不快，回去试试看 20Mhz。

griffinqiu, 2016-05-06:
@NemoAlex @shao222 汗，瀑布汗， 我一直用的40MHz。 为什么用80MHz， 会得到什么好处？

Rabbit52, 2016-06-02:
装了最新版驱动 5G 信道 149 频宽 80 传输速度稳定在 50Mb/s 左右

shao222, 2016-06-02:
换上 kmod-mwlwifi_3.18.23+10.3.0.17-20160531-1-1_mvebu.ipk 

5G Wifi 终于上 10MB/s 用几天试试效果。

bobotu, 2016-06-12:
WRT1200AC 用15.05.1 默认的驱动，5G 信道 149 频宽 80 ， 跑iperf测试结果在35-50Mb之间波动，一般是40Mb/s的样子。没试过原厂固件，不知道原厂固件是个什么状况。