NemoAlex/openwrt-wrt1900ac-docs #20: 无线驱动

griffinqiu, 2016-04-19:
看到你在[文章](http://koolshare.cn/thread-39880-1-1.html)提到”无线驱动继续使用 10.3.0.15-20151216，这个版本在 15.05.1 中表现最好“
但是为何我按照你的这个版后无线驱动显示的是10.3.0.16-20160105。

```bash
root@Griffins-WRT:~# strings /lib/modules/*.*/mwlwifi.ko | grep "^10.3"
10.3.0.16-20160105
10.3.0.16-20160105
```

griffinqiu, 2016-04-19:
是需要手动来按照10.3.0.15-20151216这个版本吗
https://github.com/NemoAlex/mwlwifi-bin

NemoAlex, 2016-04-19:
今天回去看一下，是我失误了？

Sent using CloudMagic Email [https://cloudmagic.com/k/d/mailapp?ct=pi&cv=7.4.15&pv=9.1&source=email_footer_2]

https://github.com/NemoAlex/mwlwifi-bin [https://github.com/NemoAlex/mwlwifi-bin]

NemoAlex, 2016-04-19:
确实是。1.3版不小心用了 .16 的版本。
可以尝试安装回 .15:

下载这个 `https://github.com/NemoAlex/mwlwifi-bin/blob/master/15.05.1/kmod-mwlwifi_3.18.23%2B10.3.0.15-20151216-1_mvebu.ipk`
```
opkg install --force-downgrade kmod-mwlwifi_3.18.23%2B10.3.0.15-20151216-1_mvebu.ipk
```
就可以了

closed 2016-04-19 by griffinqiu