NemoAlex/openwrt-wrt1900ac-docs #21: 我们可以实现梅林的ss游戏模式吗？

shao222, 2016-05-04:
一直有看到梅林固件的ss什么游戏模式，最近还是有v2版本。
据说对ps4 游戏，很有效。

我想了解一下，这个所谓的“游戏模式”到底是什么东西？！
和我们现在用的ss 有什么区别？

请指教！

jannson, 2016-05-04:
谢谢你的关注。我来解析下：
很多主机游戏或其它对战的游戏要求玩游戏的时候，要求有 nat2 的网络环境。（nat2 你可以搜索下，它要求 udp 内网主机从外网的nat层面打开一个端口，外网的另外一个主机能从这个外网端口访问到内网的这个主机）
而一般的 SS 只关注于 tcp，也就是看看 youtube 上上网页，服务器只管转发 tcp 而不管 udp 的转发。这个就是常常说的非游戏模式，udp 不会被服务器转发，大部分路由器是这样实现的。
或者 关注 udp 转发，但是达不到 nat2 的网络要求，游戏对战无法建立点对点的连接。
ss 在某一个版本之后，关注了游戏面临的问题，支持了 udp 的 nat2 的数据转发。这个就是常常说的游戏模式。
游戏模式v2 之前我在论坛介绍过，就是 tcp 转发兼容 原版ss，为了支持 udp in tcp与另外的 udp in udp，改了ss 的协议。所以叫v2。最大的特点是，重新用 Golang 把各个独立的程序全部重写集成在一个程序里搞定。游戏模式 v2 只需要启动一个进程，专注游戏方面的加速。（ss原版协议把 udp 数据打乱的效果很好，但游戏模式为了快，数据少，精简了些，所以目前暂时无法兼容原版协议）
游戏模式v2 目前还在探索阶段，未来希望与 ss 分开。毕竟这个东西太敏感了。所以论坛相关的介绍删除了。

shao222, 2016-05-04:
@jannson 感谢你的回复：

我现在自己用ss-redir -u 打开了 udp 的转发，这样的话对于网内的 主机游戏 是有帮助的吗？
换言之，我是不是可以理解为 ss 游戏模式约等于 ss-redir -u 模式？

jannson, 2016-05-04:
在路由器运行的时候，需要使用 tproxy 内核模块来实现udp透明转发，如果你没有用肯定不对。具体你参考下网上教程吧，很多。

houzi-, 2016-05-04:
@shao222 你需要把这两个` iptables-mod-tproxy `和 `kmod-ipt-tproxy `内核模块编译到FW里面！也可以自己接从openwrt的源安装，但是由于内核版本MD5校验肯定会造成安装失败`--force-depends`试试

shao222, 2016-05-04:
@jannson @houzi- 
我用的是 Nemo 的1.3 版本.
https://github.com/NemoAlex/openwrt-wrt1900ac-docs/releases/tag/1.3

这个应该自带tproxy 模块吧，shadowsocks 用 ss-redir -u 打开 udp 支持。
firewall 设置 tcp udp 转发。

NemoAlex, 2016-05-04:
没有自带 tproxy

shao222, 2016-05-04:
@NemoAlex 你... 

问题来了，1.我怎么添加tproxy 2. 我怎么测试udp 转发。。。

houzi-, 2016-05-04:
@shao222 很简单，输入这个命令`lsmod | grep TPROXY`该模块

shao222, 2016-05-04:
我发觉我已经装好了：

opkg list-installed |grep tproxy
iptables-mod-tproxy - 1.4.21-1
kmod-ipt-tproxy - 3.18.23-1

shao222, 2016-05-04:
firewall 也设置了转发：

iptables -t nat -A PREROUTING -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080
iptables -t nat -A OUTPUT -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080
iptables -t nat -A PREROUTING -p udp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080
iptables -t nat -A OUTPUT -p udp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080

houzi-, 2016-05-04:
我觉得你分不清楚UDP代理和UDP转发，还有你所列iptables规则也是错的！你安装了TPROXY模块还要检查一下该模块是否能正常加载！
规则写法:
https://github.com/shadowsocks/shadowsocks-libev/blob/master/README.md#advanced-usage

shao222, 2016-05-04:
能给一个比较清晰明确的方案吗？ 

我之前用的是这个 ss + ipset 的方案。
出处：https://cokebar.info/archives/962

griffinqiu, 2016-05-06:
正在下载守望先锋， 16分钟下载完。 我感觉PS4上下载速度的快慢和NAT2没多大关系。最重要的还是找一个速度快的SS。 测试方法mtr ss服务器看看掉包高不高？ 最好中间的IP是走的CN2线路

genics, 2016-07-11:
能解释下udp转发规则的写法吗?我是用的openwrt ssr客户端,已经rr-redir-u了.谢谢.
另外我们用op的真的需要一个游戏模式,请小宝考虑下.

shao222, 2016-07-11:
有个办法 利用shadowsoccks-spec 的 ss-rules 建立 udp 转发规则。
参考：https://github.com/shadowsocks/openwrt-shadowsocks/wiki/Instruction-of-ss-rules

发自我的手机