NemoAlex/openwrt-wrt1900ac-docs #25: 大家有在wrt1900ac上用 nginx 做反向代理遇到问题吗？

xdtianyu, 2016-06-20:
固件是官方的版本 

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='15.05.1'
DISTRIB_REVISION='r48532'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05.1'
DISTRIB_TAINTS=''
```

我在做 反向代理时遇到了如下错误，尝试编译 nginx-1.10.1 1.11.1 1.9.6，都遇到这个问题。

```
2016/06/20 21:23:55 [error] 3843#0: *2 upstream sent no valid HTTP/1.0 header while reading response header from upstream, client: 192.168.9.35, server: _, request: "GET /cgi-bin/luci HTTP/1.1", upstream: "http://192.168.2.120:81/cgi-bin/luci", host: "192.168.2.120"
```

参考配置就是最简单的形式

```
location /cgi-bin/luci {
    proxy_pass http://192.168.9.35;
}
```

最终在浏览器呈现为纯文本的形式， 源的 header 被直接作为内容返回了

```
Connection: close
Content-Type: text/html
Cache-Control: no-cache
Expires: 0

<!DOCTYPE html>
<html lang="zh-cn">
 <head>
  <meta charset="utf-8">
  <title>wrt1900ac - LuCI</title>
  <!--[if lt IE 9]><script src="/luci-static/bootstrap/html5.js"></script><![endif]-->
  <meta name="viewport" content="initial-scale=1.0">
  <link rel="stylesheet" href="/luci-static/bootstrap/cascade.css">
  <link rel="stylesheet" media="only screen and (max-device-width: 854px)" href="/luci-static/bootstrap/mobile.css" type="text/css" />
  <link rel="shortcut icon" href="/luci-static/bootstrap/favicon.ico">
  <script src="/luci-static/resources/xhr.js"></script>
 </head>
```

追踪了源码发现是在 `ngx_http_parse_status_line` 时出错的，也就可能是解析源返回的 `HTTP/1.1 200 OK`时出错的。

```
< HTTP/1.1 200 OK
< Connection: Keep-Alive
< Keep-Alive: timeout=20
< ETag: "1a5-1ef-56874c3f"
< Last-Modified: Sat, 02 Jan 2016 04:04:15 GMT
< Date: Mon, 20 Jun 2016 13:43:36 GMT
< Content-Type: text/html
< Content-Length: 495
```

这个可以肯定不是源返回的问题，之前使用 `ar71xx` 的固件都是正常的，而且这个配置的源是另一个 nginx 服务，配置也是最小配置，应该也不是配置问题。

leonleeann, 2017-03-09:
我也遇到这个问题，请问你最后解决了吗？怎么解决的？

xdtianyu, 2017-03-09:
@leonleeann 刷了 snapshot 版本，再没遇到这个问题

leonleeann, 2017-04-07:
那你觉得是什么原因呢？是因为哪个部分的版本更新解决了问题呢？是Nginx的版本？还是OpenWRT的？